Ankermgmt/prusaslicer-ankermake-ce-profiles #4: 256x256 PNG thumbnails are sometimes broken on the M5

neoKushan, 2023-04-17:
I've been trying out these profiles for a few days and I've noticed that sometimes the output files will refuse to load on the M5, either via USB or sending via the AnkerMake slicer. When sending via the Slicer, a generic error message is given and when trying to load the file from USB stick, it reports that the file is corrupted. 

It was fairly inconsistent too, but I noticed that it seemed to happen more on files where I'm trying to print multiple things at once. 

I didn't have these issues with the last set of profiles from @just-trey and after some investigation, I think I've figured it out - if the PNG is too large, the printer refuses to load the file at all. 

I've attached two example files, labelled GOOD and BAD - this is an example of a file I was trying to print whereby I faced the corruption. The "GOOD" File is exactly the same except I've removed the 256x256 thumbnail and this loads correctly. I've noticed that in the previously released profiles, we were using JPG as the image format rather than PNG. My hypothesis is that JPG's are smaller so we never hit whatever the threshold is that breaks the printer. It's either that or there's a bug in Anker's PNG code, but I suspect it's a file size thing as I have other files with more _objects_ in them that print fine, but the actual 256x256 PNG is much smaller. 

This is _probably_ more of a bug or a limitation in the printer itself but until Anker fixes it, I think we need to remove 256x256 thumbnails.

[Example Files.zip](https://github.com/Ankermgmt/prusaslicer-ankermake-ce-profiles/files/11249511/Example.Files.zip)

I haven't experimented with other thumbnail sizes or anything like that.

just-trey, 2023-04-17:
@neoKushan I'm testing 128x128 thumbnails exclusively. Can you try as well and let me know if it works for you on your bad example?

neoKushan, 2023-04-17:
Hey @just-trey, I did some experimenting like you asked and I have some interesting results.

I didn't like the inconsistency of the issue, so I thought I'd fudge it a little. I sliced a benchy and used that .gcode file as a base, I then modified it to have a special thumbnail in various sizes to test with.

I am convinced file size is a factor, so I created a .png file that I think would be a "worst case" scenaio. I'm not an image expert, but my understanding is that a PNG is meant to be lossless so lovely coloured gradiant like this would be incredibly inefficient and at 256x256, it comes out at 77kb:

[image]

That seems pretty "worst case" to me. 

With the test image ready to go, I then inserted it into various test versions of the .gcode file, with various sizes and combinations of sizes (in case multiple thumbnails were also a problem).

Every single version with a 256x256 thumbnail would fail to load, however version with 64x64 and 128x128 loaded just fine. For funsies, I also made some test files with 192x192 thumbnails and these also loaded fine. 

I've attached the bundle of test files
[TestFiles.zip](https://github.com/Ankermgmt/prusaslicer-ankermake-ce-profiles/files/11251084/TestFiles.zip)

Based on my findings, I think it's safe to say that 128x128 is a good safe option.

just-trey, 2023-04-17:
@neoKushan, thanks so much for testing! I am having a lot of findings in my alpha with print quality, so it will be a while, but I will use this in my next release!

commit e54b78d2 (2023-04-17): Fix thumbnail size to prevent print failure. 

Fixes #4
PR #5 (2023-04-18): Fix thumbnail size to prevent print failure.
Fixes #4
closed 2023-04-17 by just-trey