Ankermgmt/prusaslicer-ankermake-ce-profiles #9: PETG temperatures are too low

spuder, 2023-05-11:
I installed the 2.1.1 profiles, but I'm seeing that the default PETG settings are too low (`200/0`)

[image]

By contrast, the default PETG from Ankers Prusa Slicer is `240/70`

[image]

I've poked around the code, but I can't find where `200/0` comes from

just-trey, 2023-05-11:
I'll fix this and close with the next release

just-trey, 2023-05-12:
Closed in v2.2.0

closed 2023-05-12 by just-trey