Ankermgmt/prusaslicer-ankermake-ce-profiles #20: Profiles broken in prusaslicer 2.6.0-beta2 on Linux

chrivers, 2023-05-24:
Works fine for 2.5.2 on Linux.

I've installed (tried both by hand, and with the script), but 2.6.0-beta2 fails. This is the startup message:

```
[2023-05-24 23:00:45.095508] [0x00007f409ccbb880] [trace]   Initializing StaticPrintConfigs
[2023-05-24 23:00:45.399221] [0x00007f409ccbb880] [error]   flatten_configbundle_hierarchy: The preset 0.25mm DRAFT @ANKER-CE inherits an unknown preset ""
```

I think that error is from this trailing semicolong:

```
inherits = *0.25mm*;
```

the surrounding lines don't have it, and when I remove it, the error message goes away.

However, even with this change, I can't see the profiles in beta2 :(

Any ideas?

chrivers, 2023-05-24:
The option just doesn't show up in the config wizard. But in non-beta 2.5, it shows up as expected.

chrivers, 2023-05-24:
I found a clue. When opening the config wizard, the following log appears:

```
[error]   Could not load bundle AnkerMakeCE due to missing index ~/.config/PrusaSlicer-beta/cache/AnkerMakeCE.idx.
[error]   Profile Generic PET @ANKER was not found in installed vendor Preset Bundles.
```

I'm not sure if the `Generic PET @ANKER` is a real issue, but maybe?

However, after I manually copied `AnkerMakeCE.idx` into `cache`, the profile can be loaded:

```sh
cd ~/.config/PrusaSlicer-beta
cp vendor/AnkerMakeCE.idx cache
```

This might be a problem with prusaslicer, more than the profiles?

just-trey, 2023-05-24:
@chrivers Talking with Prusa about this over  here. They have some suggestions but it's still a bit sketchy https://github.com/prusa3d/PrusaSlicer/issues/10606

just-trey, 2023-06-07:
This is resolved in 2.6.0-beta 4

closed 2023-06-07 by just-trey